vincentwolsink/home_assistant_enphase_envoy_installer #167: Entities for firmware version

M66B, 2024-10-28:
**Describe the feature**
Entity for Envoy firmware version, and micro inverters firmware versions.
Currently, the firmware version is shown for a device, but there is no entity, which would be useful to monitor firmware updates.

**Account type**
- [X] Installer
- [ ] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [ ] Metered with CTs installed
- [ ] Metered without CTs
- [X] Standard

- FW version: D8.2.4225 (this is what it is about)
- Amount of micro inverters connected: 10

**Home Assistant**
- Version: 2024.10.3

mnederlof, 2024-10-28:
Great feature, but only useful, if we know how to detect if a firmware update is available and how to trigger it.

Do you have any information on how that can be found / done?

mnederlof, 2024-10-28:
but how do we know if there is a firmware update available? And also, the firmware is pushed by enphase, we as end-users/installers have very limited influence on that.

So in which api should we look for a firmware version that is available for the envoy in order to show that an update is even available?

vincentwolsink, 2024-10-28:
I think @M66B wants to use it to know that (and when) the Envoy has automatically updated. Not if an update is available.

M66B, 2024-10-28:
> I think @M66B wants to use it to know that (and when) the Envoy has automatically updated. Not if an update is available.

Indeed.

vincentwolsink, 2024-10-28:
@M66B The sensors are available in [0.5.1](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/releases/tag/0.5.1)

closed 2024-10-28 by vincentwolsink

M66B, 2024-10-28:
@vincentwolsink Thanks. However, I use the master branch, living dangerously ;-)