vincentwolsink/home_assistant_enphase_envoy_installer #170: [FEATURE] Production switch on standard gateway

remcodk, 2024-11-03:
**Describe the feature**
The production switch is not visual. Maybe because I have standard Envoy. I would like to use the switch production option.

**Account type**
- [x] Installer
- [ ] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [ ] Metered with CTs installed
- [ ] Metered without CTs
- [x] Standard

- FW version: D8.2.4264
- Amount of micro inverters connected: 14

**Home Assistant**
- Version: 2024.10

vincentwolsink, 2024-11-04:
Hi @remcodk, the switch should be available also on Envoy Standard. Can you please turn on debug logging, reload the integration and provide the debug log?

remcodk, 2024-11-04:
I have created the log. What is a good way to get the file to you? There are serialnumbers and tokens in it so maybe not the best solution to paste it here.

vincentwolsink, 2024-11-04:
You can send it to temporary e-mail ziener06joduur@icloud.com

vincentwolsink, 2024-11-04:
It seems the `production_power` endpoint is disabled in the integration configuration. Did you do that yourself? Try enabling it again.

remcodk, 2024-11-04:
Indeed I disabled it to test if that changes the behavior. I enabled the endpoints and sent a new log to you.

Maybe this is the cause:
2024-11-04 13:36:19.085 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] Resolving jsonpath endpoint_production_power.powerForcedOff
2024-11-04 13:36:19.085 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] the configured path endpoint_production_power.powerForcedOff did not return anything!

vincentwolsink, 2024-11-04:
It is skipping the endpoint because an installer token is required and the login you provided is home owner only. Are you sure your login has installer access to the site?

remcodk, 2024-11-04:
When I browse to my gateway I have access to the installer page. Maybe my current token in Home Assistant is incorrect?

[image]

vincentwolsink, 2024-11-04:
Do you use a separate installer company login or do you have a DHZ type account for your own login?

remcodk, 2024-11-04:
No I created only one login at Enphase. And installed the solar system myself. I can't remember if I chose DHZ or installer account type.

remcodk, 2024-11-04:
I retrieved a token with https://entrez.enphaseenergy.com/ and it now gives "installer" instead of "owner". Is there a simple way to update the token in Home Assistant?

"enphaseUser": "installer",

remcodk, 2024-11-04:
Deleted all devices and logged in again. It is working now! Thanks for your time.

[image]