vincentwolsink/home_assistant_enphase_envoy_installer #171: [BUG] Error doing job: Task exception was never retrieved (None)

Texass12, 2024-11-10:
Hello community,
I get the following error report in the log, maybe someone can help me here. I'm not particularly good at reading out whether it's the software or my system.
I didn't notice any errors like this in the log before the update.

**Describe the bug**
A clear and concise description of what the bug is.

**Account type**
- [ ] Installer
- [ ] DIY / DHZ
- [x] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [x] Metered with CTs installed
- [ ] Metered without CTs
- [ ] Standard

- FW version: D7.6.336
- Amount of micro inverters connected: 46

**Home Assistant**
- Version: 2024.11.1

**Additional context**
```
Relevant snippet of Home Assistant error log. 

Logger: homeassistant
Quelle: custom_components/enphase_envoy/sensor.py:380
Integration: Enphase Envoy (Installer) (Dokumentation, Probleme)
Erstmals aufgetreten: 10:47:16 (46 Vorkommnisse)
Zuletzt protokolliert: 11:36:14

Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 266, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 481, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 176, in async_update_listeners
    update_callback()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 561, in _handle_coordinator_update
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1007, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1132, in _async_write_ha_state
    self.__async_calculate_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1073, in __async_calculate_state
    if extra_state_attributes := self.extra_state_attributes:
                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/sensor.py", line 380, in extra_state_attributes
    .get("report_date")
     ^^^
AttributeError: 'NoneType' object has no attribute 'get'

```

vincentwolsink, 2024-11-11:
Hi @Texass12, could you please download the diagnostic file for this integration and paste it here (or e-mail to temporary email ziener06joduur@icloud.com).

vincentwolsink, 2024-11-11:
Should be fixed in [0.5.3](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/releases/tag/0.5.3)