vincentwolsink/home_assistant_enphase_envoy_installer #183: [BUG] Failed setup, will retry: 'acVoltageINmV'

Kudos50, 2024-11-30:
After upgrading to 0.6.0 or any of its beta's I am unable to connect to envoy S (non-metered). It is no longer able to connect and removing/re-adding gives the same error.

For now I downgraded to 0.5.3 without issues. All sensors back online for both Envoys.

I use discovery so I'm only typing the installer email address and password:
<CUT>
2024-11-30 11:29:31.855 INFO (MainThread) [custom_components.enphase_envoy.envoy_reader] TOKEN TYPE: installer
2024-11-30 11:29:32.123 INFO (MainThread) [custom_components.enphase_envoy.envoy_reader] Updating endpoints ['endpoint_production_json']
2024-11-30 11:29:32.123 INFO (MainThread) [custom_components.enphase_envoy.envoy_reader] UPDATING ENDPOINT endpoint_production_json: https://{}/production.json?details=1
2024-11-30 11:29:32.961 INFO (MainThread) [custom_components.enphase_envoy.envoy_reader] - FETCHING ENDPOINT endpoint_production_json TOOK 0.8377 seconds
2024-11-30 11:29:32.961 INFO (MainThread) [custom_components.enphase_envoy.envoy_reader] Updating endpoints ['endpoint_production_v1']
2024-11-30 11:29:32.962 INFO (MainThread) [custom_components.enphase_envoy.envoy_reader] UPDATING ENDPOINT endpoint_production_v1: https://{}/api/v1/production
2024-11-30 11:29:33.194 INFO (MainThread) [custom_components.enphase_envoy.envoy_reader] - FETCHING ENDPOINT endpoint_production_v1 TOOK 0.2326 seconds
2024-11-30 11:29:33.196 INFO (MainThread) [custom_components.enphase_envoy.envoy_reader] Updating endpoints {'endpoint_admin_tariff', 'endpoint_info', 'endpoint_ensemble_inventory', 'endpoint_ensemble_power', 'endpoint_device_data', 'endpoint_production_json', 'endpoint_production_power', 'endpoint_ensemble_secctrl', 'endpoint_pdm_energy', 'endpoint_installer_agf', 'endpoint_pcu_comm_check', 'endpoint_inventory'}
2024-11-30 11:29:33.196 INFO (MainThread) [custom_components.enphase_envoy.envoy_reader] UPDATING ENDPOINT endpoint_admin_tariff: https://{}/admin/lib/tariff
2024-11-30 11:29:33.468 INFO (MainThread) [custom_components.enphase_envoy.envoy_reader] - FETCHING ENDPOINT endpoint_admin_tariff TOOK 0.2713 seconds
2024-11-30 11:29:33.469 INFO (MainThread) [custom_components.enphase_envoy.envoy_reader] UPDATING ENDPOINT endpoint_info: https://{}/info.xml
2024-11-30 11:29:34.454 INFO (MainThread) [custom_components.enphase_envoy.envoy_reader] - FETCHING ENDPOINT endpoint_info TOOK 0.9851 seconds
2024-11-30 11:29:34.457 INFO (MainThread) [custom_components.enphase_envoy.envoy_reader] UPDATING ENDPOINT endpoint_ensemble_inventory: https://{}/ivp/ensemble/inventory
2024-11-30 11:29:34.688 INFO (MainThread) [custom_components.enphase_envoy.envoy_reader] - FETCHING ENDPOINT endpoint_ensemble_inventory TOOK 0.2302 seconds
2024-11-30 11:29:34.688 INFO (MainThread) [custom_components.enphase_envoy.envoy_reader] UPDATING ENDPOINT endpoint_ensemble_power: https://{}/ivp/ensemble/power
2024-11-30 11:29:34.931 INFO (MainThread) [custom_components.enphase_envoy.envoy_reader] - FETCHING ENDPOINT endpoint_ensemble_power TOOK 0.2425 seconds
2024-11-30 11:29:34.931 INFO (MainThread) [custom_components.enphase_envoy.envoy_reader] UPDATING ENDPOINT endpoint_device_data: https://{}/ivp/pdm/device_data
2024-11-30 11:29:35.176 INFO (MainThread) [custom_components.enphase_envoy.envoy_reader] - FETCHING ENDPOINT endpoint_device_data TOOK 0.2445 seconds
2024-11-30 11:29:35.179 ERROR (MainThread) [custom_components.enphase_envoy] Unexpected error fetching envoy Envoy 122226007074 data
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 382, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 279, in _async_update_data
    return await self.update_method()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/__init__.py", line 102, in async_update_data
    await envoy_reader.get_data()
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 1111, in get_data
    await self.update_endpoints()  # fetch all remaining endpoints
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 1095, in update_endpoints
    self.data.set_endpoint_data(endpoint, getattr(self, endpoint))
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 271, in set_endpoint_data
    self.data[endpoint] = parse_devicedata(response.json())
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 84, in parse_devicedata
    "ac_voltage": int(last_reading["acVoltageINmV"]) / 1000,
                      ~~~~~~~~~~~~^^^^^^^^^^^^^^^^^
KeyError: 'acVoltageINmV'
<PASTE>

vincentwolsink, 2024-11-30:
Hi! Thanks for the log output. Can you please login to your local Envoy interface and send the output of https://envoy.local/ivp/pdm/device_data

Kudos50, 2024-11-30:
Needed to add .txt extension to upload. Also, please note that envoy.local is not really working for me as I have 2 units on the local network.
[device_data.txt](https://github.com/user-attachments/files/17966736/device_data.txt)

closed 2024-12-01 by vincentwolsink

Kudos50, 2024-12-01:
Confirmed. Thanks Vincent!

PR #184 (2024-12-01): Only read device_data for active devices
Fixes #183