vincentwolsink/home_assistant_enphase_envoy_installer #185: [BUG] KeyError: 'VrmsL1N'

sjhilton1978, 2024-12-01:
**Describe the bug**
Error on restart and configuration will not complete for v0.6.0 or 0.6.1 with error KeyError: 'VrmsL1N' recorded in the log

**Account type**
- Installer

**Envoy**
- Metered with CTs installed

- FW version: D7.6.168 (f1975b)
- Amount of micro inverters connected: 24

**Home Assistant**
- Version: 2024.11.3

**Additional context**
```
Relevant snippet of Home Assistant error log. 
2024-12-01 21:48:40.232 ERROR (MainThread) [custom_components.enphase_envoy] Unexpected error fetching envoy Envoy 122404002310 data
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 382, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 279, in _async_update_data
    return await self.update_method()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/__init__.py", line 102, in async_update_data
    await envoy_reader.get_data()
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 1115, in get_data
    await self.update_endpoints()  # fetch all remaining endpoints
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 1099, in update_endpoints
    self.data.set_endpoint_data(endpoint, getattr(self, endpoint))
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 275, in set_endpoint_data
    self.data[endpoint] = parse_devicedata(response.json())
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 114, in parse_devicedata
    "voltage_l1": int(last_reading["VrmsL1N"]) / 1000,
                      ~~~~~~~~~~~~^^^^^^^^^^^
KeyError: 'VrmsL1N'

2024-12-01 21:49:29.447 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry Envoy [Serial No redacted] for enphase_envoy
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 635, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/__init__.py", line 125, in async_setup_entry
    await coordinator.async_config_entry_first_refresh()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 309, in async_config_entry_first_refresh
    await self._async_refresh(
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 382, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 279, in _async_update_data
    return await self.update_method()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/__init__.py", line 102, in async_update_data
    await envoy_reader.get_data()
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 1115, in get_data
    _LOGGER.info(
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 1088, in update_endpoints
    if endpoint_settings == None:
        ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 719, in _update_endpoint
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 741, in _async_fetch_with_retry
    formatted_url = url.format(self.host)
                   ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/httpx/_client.py", line 1814, in get
    return await self.request(
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/httpx/_client.py", line 1585, in request
    return await self.send(request, auth=auth, follow_redirects=follow_redirects)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/httpx/_client.py", line 1674, in send
    response = await self._send_handling_auth(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/httpx/_client.py", line 1702, in _send_handling_auth
    response = await self._send_handling_redirects(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/httpx/_client.py", line 1739, in _send_handling_redirects
    response = await self._send_single_request(request)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/httpx/_client.py", line 1776, in _send_single_request
    response = await transport.handle_async_request(request)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/httpx/_transports/default.py", line 377, in handle_async_request
    resp = await self._pool.handle_async_request(req)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/httpcore/_async/connection_pool.py", line 216, in handle_async_request
    raise exc from None
  File "/usr/local/lib/python3.12/site-packages/httpcore/_async/connection_pool.py", line 196, in handle_async_request
    response = await connection.handle_async_request(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/httpcore/_async/connection.py", line 101, in handle_async_request
    return await self._connection.handle_async_request(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/httpcore/_async/http11.py", line 143, in handle_async_request
    raise exc
  File "/usr/local/lib/python3.12/site-packages/httpcore/_async/http11.py", line 113, in handle_async_request
    ) = await self._receive_response_headers(**kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/httpcore/_async/http11.py", line 186, in _receive_response_headers
    event = await self._receive_event(timeout=timeout)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/httpcore/_async/http11.py", line 224, in _receive_event
    data = await self._network_stream.read(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/httpcore/_backends/anyio.py", line 35, in read
    return await self._stream.receive(max_bytes=max_bytes)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/anyio/streams/tls.py", line 204, in receive
    data = await self._call_sslobject_method(self._ssl_object.read, max_bytes)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/anyio/streams/tls.py", line 147, in _call_sslobject_method
    data = await self.transport_stream.receive()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/anyio/_backends/_asyncio.py", line 1227, in receive
    await self._protocol.read_event.wait()
  File "/usr/local/lib/python3.12/asyncio/locks.py", line 212, in wait
    await fut
asyncio.exceptions.CancelledError: Home Assistant is stopping

chicknwinner, 2024-12-01:
also experiencing this issue.

commented out these lines in envoy_reader.py and it works ok

                "voltage_l1": int(last_reading["VrmsL1N"]) / 1000,
                "voltage_l2": int(last_reading["VrmsL2N"]) / 1000,
                "voltage_l3": int(last_reading["VrmsL3N"]) / 1000,

also tiny thing noticed that for 0.6.1 the version wasn't changed from 0.6.0 in custom_components/enphase_envoy/manifest.json

vincentwolsink, 2024-12-01:
> also experiencing this issue.
>
> commented out these lines in envoy_reader.py and it works ok
>
> ```
>             "voltage_l1": int(last_reading["VrmsL1N"]) / 1000,
>             "voltage_l2": [...]

Ah, indeed. Very early morning 😅

Could one of you please send the output of https://envoy.local/ivp/pdm/device_data (after logging in)

chicknwinner, 2024-12-01:
[device_data.txt](https://github.com/user-attachments/files/17968864/device_data.txt)

vincentwolsink, 2024-12-02:
Should be better in 0.6.2.

closed 2024-12-02 by vincentwolsink