vincentwolsink/home_assistant_enphase_envoy_installer #187: [BUG] did i do something wrong?

Seaniemo, 2024-12-01:
**Describe the bug**
i see the update with the tip that HA creates new entity's with the same name but different ID's so its adviced to remove and readd the enphase enoy (installer) device to my HA. now i did this and some entities are not there anymore. i can nowhere find 
sensor.envoy_xxx_current_power_production for example. 

did i do something wrong.?

**Account type**
- [ ] Installer
- [ X] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [ ] Metered with CTs installed
- [ ] Metered without CTs
- [ X] Standard

- FW version: D8.2,4264
- Amount of micro inverters connected: 9

**Home Assistant**
- Version: 2024.11.3

**Additional context**
``` i re added the integration when  no sun is shining 
Relevant snippet of Home Assistant error log. 
```

TGgrimreaper, 2024-12-01:
Ha renamed it to sensor.envoy_xxx_current_power_production_2.

So long term stats is lost🥲

mnederlof, 2024-12-01:
You can probably just rename the sensor name to without the `_2` and see your long term stats.

TGgrimreaper, 2024-12-01:
Not working for me. Says already registered. Changed the old one to _1 and the new one removed _2 still same

vincentwolsink, 2024-12-01:
You can rename and preserve your long term stats. Make sure the old sensor is deleted and remove the _2 from the new one. If that doesn’t work restart Home Assistant and try again.

Seaniemo, 2024-12-01:
> You can rename and preserve your long term stats. Make sure the old sensor is deleted and remove the _2 from the new one. If that doesn’t work restart Home Assistant and try again.

Could you give me a step by step ? Because what I did ain't working. 

Would be really appreciated.

vincentwolsink, 2024-12-02:
1. Remove your Envoy from the integration.
2. Restart Home Assistant.
3. Add your Envoy to the integration.

Should be this simple.

Seaniemo, 2024-12-02:
> 1. Remove your Envoy from the integration.
> 2. Restart Home Assistant.
> 3. Add your Envoy to the integration.
> 
> Should be this simple.

Vincent, what i did was the following. 
1: update the integration
2: remove the integration
3: restart home assitant
4: add the integration. 

and for some reason this did not working and couldnt find current_power no more. 

should i not update ?

vincentwolsink, 2024-12-02:
Can you please provide me with the diagnostics file from the integration?

Seaniemo, 2024-12-02:
[config_entry-enphase_envoy-01JE480B6Q3WDT0D0PXAM6FRJK.json](https://github.com/user-attachments/files/17982081/config_entry-enphase_envoy-01JE480B6Q3WDT0D0PXAM6FRJK.json)
@vincentwolsink here is the diagnostics. i will restore my system again now. 

[image]
here you can see no more power production entity

Seaniemo, 2024-12-02:
@vincentwolsink i just updated without removing. and now it just works. 

i will just delete the old un available entities...

closed 2024-12-03 by vincentwolsink