vincentwolsink/home_assistant_enphase_envoy_installer #189: [BUG] Missing inverter current production value on 0.6.x

arnauos, 2024-12-02:
Hi,

After upgrading to 0.6.x all my inverters are missing the current production value (in watts). 

This was previously available on a sensor for each inverter called: sensor.envoy_XXXXXXXXXXXX_inverter_YYYYYYYYYY

After upgrading I only see for each inverter:

- Communicating (binary: ON / OFF)
- Firmware Version (this is the only one that got duplicated)
- Producing (binary: ON / OFF)

**Account type**
- [ ] Installer
- [ ] DIY / DHZ
- [X] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [X] Metered with CTs installed
- [ ] Metered without CTs
- [ ] Standard

- FW version: D7.6.175
- Amount of micro inverters connected: 10

**Home Assistant**
- Version: 2024.11.3

vincentwolsink, 2024-12-02:
Hi! Could you please send me the output of https://envoy.local/ivp/pdm/device_data (after logging in)

arnauos, 2024-12-02:
> Hi! Could you please send me the output of https://envoy.local/ivp/pdm/device_data (after logging in)

I'm not sure on how to get to there, after login if I open that url it says "401 auth required" and brings me back to /home.

vincentwolsink, 2024-12-02:
Did you “Login with Enphase” on the home page? Because in that case it could be the new endpoint we are using is not available on firmware 7 yet 😓

arnauos, 2024-12-02:
Yes... that may be the problem.

vincentwolsink, 2024-12-02:
Just to be sure. Could you enable debug logging for the integration, reload the integration and then send me (on temporary e-mail ziener06joduur@icloud.com) the log output?

arnauos, 2024-12-02:
Sent! (I do see the 401 for the endpoint in there)

vincentwolsink, 2024-12-03:
Should be fixed in 0.6.3.

closed 2024-12-03 by vincentwolsink

arnauos, 2024-12-03:
Can confirm. Sensor appeared with a new name (now sensor.inverter_xxx_production vs sensor.envoy_xxx_inverter_yyy_production) but it is correctly reporting production value... going to rename them now to keep the long term stats

Thanks!!