vincentwolsink/home_assistant_enphase_envoy_installer #191: [BUG] Missing many data for some of the inverters after upgrading to 0.6.x

upsuper, 2024-12-07:
**Describe the bug**
After upgrading to 0.6.x, most inverters are working fine (albeit I need to remove some old sensors and update the ID of the new sensors), but there are some inverters lose majority of its sensors. See the comparison below:

<details>
<summary>Screenshots showing different inverters</summary>

| Inverter that works normally | Inverter that loses majority of the sensors |
| --- | --- |
| [image] | [image] |

</details>

**Account type**
- [ ] Installer
- [x] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [x] Metered with CTs installed
- [ ] Metered without CTs
- [ ] Standard

- FW version: D8.2.127
- Amount of micro inverters connected: 56

**Home Assistant**
- Version: 2024.12.1

**Additional context**

I took a look at some other issues, specially I saw https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/issues/189#issuecomment-2512484180 and also tried to fetch data from `ivp/pdm/device_data`, and it seems that it only lists 50 inverters from that endpoint, and the inverters lose the sensors can't be found in that data. So it seems like this endpoint doesn't work as reliably as the previous one I suppose?

upsuper, 2024-12-07:
And indeed I can found the inverters in the previous `ivp/peb/devstatus` endpoint but not the new one.

vincentwolsink, 2024-12-07:
Hi! That is a serious amount of inverters 😁

Can you please provide the output of the `ivp/pdm/device_data` endpoint?

catsmanac, 2024-12-07:
Looks like the tail-end of device_data shows the limit of 50. Device count matches my 24 inverters and 1 q-relay.

```json
    "deviceCount": 25,
    "deviceDataLimit": 50
```

upsuper, 2024-12-07:
Oh, good point. I have 56 inverters + 2 relays so it should be 58 devices in total. Maybe that's over the limit of this endpoint?

catsmanac, 2024-12-07:
Might be something that can be configured as datasheets states it can communicate with up to 300 inverters.

upsuper, 2024-12-07:
Maybe @krbaker can help shed some light on how to use a different limit or do the pagination on this endpoint?

vincentwolsink, 2024-12-07:
Maybe this limit is configurable. Or they implemented some sort of pagination.

vincentwolsink, 2024-12-09:
For now, what I can do is if there are more than 50 devices, fallback to the `devstatus` endpoint. This is not ideal though. Better is to find out how to do pagination..

vincentwolsink, 2024-12-11:
Things are now getting a bit too complex for my liking. But I made usage of the old `devstatus` endpoint configurable for now in #195 .

vincentwolsink, 2024-12-11:
Keeping this open, since pagination on the new endpoint would still be the preferred solution.

upsuper, 2024-12-13:
I can confirm that enabling `devstatus` endpoint option in configuration and reload the integration makes those information come back, thanks! (It does mean all the new sensors become unavailable but that's okay.)

closed 2026-09-04 by vincentwolsink