vincentwolsink/home_assistant_enphase_envoy_installer #192: [BUG] Using deprecated constant (as per 2025.9)

KomtGoed, 2024-12-07:
**Describe the bug**

See the following line appearing in my logs:
`2024-12-07 05:03:23.571 WARNING (ImportExecutor_0) [homeassistant.const] POWER_VOLT_AMPERE_REACTIVE was used from enphase_envoy, this is a deprecated constant which will be removed in HA Core 2025.9. Use UnitOfReactivePower.VOLT_AMPERE_REACTIVE instead, please report it to the author of the 'enphase_envoy' custom integration `

**Account type**
- [ ] Installer
- [ ] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [ ] Standard

- FW version: D7.xxx
- Amount of micro inverters connected: x

**Home Assistant**
- Version: 2024.11

vincentwolsink, 2024-12-07:
Hi! This was already fixed in #152, included in version 0.4.5. Are you sure you are running the latest version?

KomtGoed, 2024-12-08:
Sorry for the confusion (and using your time). I was indeed stuck on 0.4.3 for whatever reason, didn't get an "update available"-notification... Note to self: next time check version first.

On 0.6.3 now, message disappeared from log files. Thanks!

closed 2024-12-08 by KomtGoed