vincentwolsink/home_assistant_enphase_envoy_installer #194: [BUG] no realtime data, energy dashboard not updating

TGgrimreaper, 2024-12-11:
**Describe the bug**
A clear and concise description of what the bug is.

**Account type**
- [ ] Installer
- [x ] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [ x] Metered with CTs installed
- [ ] Metered without CTs
- [ ] Standard

- FW version: D7.xxx
- Amount of micro inverters connected: 12

**Home Assistant**
- Version: 2024.12.2

**Additional context**
``` since last night (01.30.00 the data is not updating
Relevant snippet of Home Assistant error log. 
```
Deze fout is ontstaan door een aangepaste integratie.

Logger: custom_components.enphase_envoy.envoy_reader
Bron: custom_components/enphase_envoy/envoy_reader.py:1053
integratie: Enphase Envoy (Installer) (documentatie, problemen)
Eerst voorgekomen: 19:16:57 (1 gebeurtenissen)
Laatst gelogd: 19:16:57

Stopped reading realtime data

vincentwolsink, 2024-12-11:
Hi! Realtime data needs a stable (tcp) connection to your Envoy. Is there an error before the `Stopped reading realtime data`? That should show the actual reason for stopping.

TGgrimreaper, 2024-12-11:
The envoy (Metered fw D8.2.4264) is connected to the internet via lan. I can see the data in the official enphase app. So that us fine. By the looks of it the sensors still update but the graphics not...
[image]
[image]

vincentwolsink, 2024-12-11:
Is there an error before the `Stopped reading realtime data`? That should show the actual reason for stopping.

TGgrimreaper, 2024-12-11:
Detected that custom integration 'enphase_envoy' sets option flow config_entry explicitly, which is deprecated at custom_components/enphase_envoy/config_flow.py, line 232: self.config_entry = config_entry. This will stop working in Home Assistant 2025.12, please create a bug report at
Logger: homeassistant.helpers.frame
Bron: helpers/frame.py:324
Eerst voorgekomen: 19:16:38 (1 gebeurtenis)
Laatst gelogd: 19:37:22

2024-12-11 18:12:29.047 ERROR (MainThread) [custom_components.enphase_envoy.envoy_reader] Stopped reading realtime data

2024-12-11 18:12:29.047 WARNING (MainThread) [custom_components.enphase_envoy] Re-connecting /stream/meter

closed 2025-02-24 by vincentwolsink