vincentwolsink/home_assistant_enphase_envoy_installer #197: [BUG] Integration doesn’t work with core 2025.1.0

Robsta86, 2025-01-05:
**Describe the bug**
After updating core to 2025.1.0 the integration stops working.

**Account type**
- [ ] Installer
- [ ] DIY / DHZ
- [X] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [X] Metered with CTs installed
- [ ] Metered without CTs
- [ ] Standard

**Home Assistant**
- Version: 2025.1.0

vincentwolsink, 2025-01-06:
Hi. There should be no issues running this integration with 2025.1.0. Try reloading it. Or please post relevant errors from the system error log.

Andre-vdt, 2025-01-10:
Hello there. Any news on this matter? Same Happening here. Unable to get Enphase Integration working with the 2025.x Core updates.

mnederlof, 2025-01-10:
It is complaining about line 13 of [enphase_envoy/const.py](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/blob/main/custom_components/enphase_envoy/const.py#L13), but that does not match with the current master version.

Which version of this integration are you running?

I can confirm that i am running version 0.6.3 with HA version 2025.2 without issues.

mnederlof, 2025-01-11:
so upgrading to latest version fixed it according to @Andre-vdt 😄

closed 2025-01-11 by mnederlof

bidinaute, 2025-01-22:
> Hello there. Any news on this matter? Same Happening here. Unable to get Enphase Integration working with the 2025.x Core updates.
> 
> Mea Culpa Old version in place. Downloaded 0.6.4 Problem solved.

Hello @Andre-vdt could you explain to me how you switched to 0.6.4 ? I currently use the 0.1.2 and I can't remove or upgrade it for the moment.

Many thanks

B.