vincentwolsink/home_assistant_enphase_envoy_installer #202: Battery Capacity issue

minggaard, 2025-02-05:
I have a system with 3 * Encharge 3T.
Which is supposed to be at 3.5 kWh each.
But looking at the capacity in this HA integration its reported as 4960 Wh each.
also the total Batteries Capacity reported by the Envoy is 14880 Wh
Not that I'm unhappy with additional capacity but guess its not correct.
Any idea of why this occur ?

Account type

        Installer
        DIY / DHZ
        [x ] Home Owner (This integration is not (fully) functional with a home owner account!)

Envoy

        [ x] Metered with CTs installed

        Metered without CTs

        Standard

        FW version: D8.2.4382

        Amount of micro inverters connected: 36

Home Assistant

    Version: 2025.1.4

vincentwolsink, 2025-02-12:
Hi @minggaard, no clue why the Envoy would report that. Just to be sure, could you attach the output of https://envoy.local/ivp/ensemble/inventory (you need to login)?

vincentwolsink, 2025-02-24:
The Envoy indeed reports that capacity. I have no clue why it would do that. Nothing I can do about it here :)

closed 2025-02-24 by vincentwolsink