vincentwolsink/home_assistant_enphase_envoy_installer #205: Depreciated constant needs reconfiguration with updated constant.

ARTSRAY, 2025-02-17:
From the logs

Logger: homeassistant.const
Source: helpers/deprecation.py:222
First occurred: February 14, 2025 at 20:19:46 (1 occurrences)
Last logged: February 14, 2025 at 20:19:46

POWER_VOLT_AMPERE_REACTIVE was used from enphase_envoy, this is a deprecated constant which will be removed in HA Core 2025.9. Use UnitOfReactivePower.VOLT_AMPERE_REACTIVE instead, **please report it to the author of the 'enphase_envoy' custom integration**

vincentwolsink, 2025-02-17:
You are probably running an older version of this integration. Try updating to the latest.

ARTSRAY, 2025-02-17:
I have 0.3.8 currently. I was expecting the integration would update automatically?

vincentwolsink, 2025-02-17:
Custom integrations normally don’t auto update. If you used [HACS](https://www.hacs.xyz/) to install it will notify you of a new version and can do the update for you.

If you manually installed, you need to do the updates yourself.

ARTSRAY, 2025-02-18:
Hi Vincent, I am trying to reinstall the new version. I removed the old version first. 

I now receive an error: "Unexpected error" after adding the IP address, Serial number and Enlighten user ID. Anyway you can help me resolve this? Do I need to create a API token first?

ARTSRAY, 2025-02-18:
This error originated from a custom integration.

Logger: custom_components.enphase_envoy.config_flow
Source: custom_components/enphase_envoy/config_flow.py:181
integration: Enphase Envoy (Installer) (documentation, issues)
First occurred: 08:29:05 (4 occurrences)
Last logged: 10:06:00

Unexpected exception
Traceback (most recent call last):
  File "/config/custom_components/enphase_envoy/config_flow.py", line 181, in async_step_user
    envoy_reader = await validate_input(self.hass, user_input)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/config_flow.py", line 54, in validate_input
    await envoy_reader.get_data()
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 1194, in get_data
    await self.init_authentication()
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 1041, in init_authentication
    await self._get_enphase_token()
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 963, in _get_enphase_token
    self._token = await self._fetch_envoy_token_json()
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 933, in _fetch_envoy_token_json
    raise EnlightenError("Login did not succeed")
custom_components.enphase_envoy.envoy_reader.EnlightenError: Login did not succeed

ARTSRAY, 2025-02-18:
Vincent, I made a typo in the password section, now it loads successfully.
Thank you.