vincentwolsink/home_assistant_enphase_envoy_installer #207: [BUG] Enphase Envoy (Installer) integration cannot be installed / used in parallel to 'official'/core Enphase Envoy one

joergbattermann, 2025-02-19:
**Describe the bug**
I've been using the official, [core Enphase Envoy](https://www.home-assistant.io/integrations/enphase_envoy/) integration for a while and found out about [this one](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer) a little while ago. I was curious and wanted to give it a try but even though I _could_ add & download it via HACS, after the HA restart the core integration was broken and could not be loaded anymore and I _think_ it's because both integrations use the same domain (see [this](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/blob/main/custom_components/enphase_envoy/manifest.json#L2) and [this](https://github.com/home-assistant/core/blob/dev/homeassistant/components/enphase_envoy/manifest.json#L2)).

Historically it appears like this one here [had a different / *_custom domain](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/blame/ee23a059ff7c1edb77bd2be6ed1d6255bc24e20d/custom_components/enphase_envoy_custom/manifest.json) but was changed a few commits in so I was wondering if this is known/intentional and if there are any plans to make both integrations to work in parallel given that they do have slightly different functionalities etc?

Thanks!

**Account type**
- [ ] Installer
- [x] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [x] Metered with CTs installed
- [ ] Metered without CTs
- [ ] Standard

- FW version: D7.xxx
- Amount of micro inverters connected: 35

**Home Assistant**
- Version: 2025.2.x

**Additional context**
```
Relevant snippet of Home Assistant error log. 
```

vincentwolsink, 2025-02-21:
Hi! Correct. This was an unlucky design choice in the past. This integration used to be a drop-in replacement with extra functionality. But nowadays the official integration changed a lot and both are no longer compatible.

You would need to remove the configuration for the official integration first and then you can install and configure this one.

joergbattermann, 2025-02-21:
Thanks for the quick response @vincentwolsink .. appreciate it!

Is it realistic to keep this issue around with the hope / advocate for this domain / name overlap to being changed? I understand that this would be a (significant) breaking change and I'd certainly not expect it to happen quickly and maybe more akin of announcing it a few times in the next versions' release notes.. but ideally at one point one could just install both (& without tinkering etc).

vincentwolsink, 2025-02-24:
Yes, I would like to move the domain at some point. I wonder how much chaos it will cause though 😅

joergbattermann, 2025-02-24:
I mean.. what _COULD_ go wrong (other than dashboards and custom templates and all that breaking) ;-) ?

But yeah, that's why I sort of meant.. let people know a couple of versions upfront in each release note leading up to i.e.  v0.**7**.0 or v**1**.0 or sth and then everyone got enough time to mentally prepare / hold back updating until they have a little bit of time to adjust their customizations around the current naming scheme.

bartkummel, 2025-04-05:
I have a similar issue: I'm using [the other HACS Enphase integration](https://github.com/briancmpbll/home_assistant_custom_envoy). But now I want to be able to switch production on and off, so I want to switch to this integration. I also don't want to loose all data that I've collected about energy production in the past. 

- Would I need to wait for this integration to change it's domain before I can install it without uninstalling the other one?
- Or can I keep my old data when I uninstall the old integration and then install this one?

vincentwolsink, 2025-04-06:
Hi @bartkummel, Home Assistant, does not delete history when you remove an integration.

You can remove the previous one and then install this one. If you make sure the (important) sensors have the same name, history will be there.

jvrossum, 2025-05-03:
I downloaded and configured this addon, but it does not start. But the “official” integration is nowhere to be found. So now my old integration is not longer working and your addon is also not working.. The official add on does not show either in the integration list anymore..

vincentwolsink, 2025-05-03:
@jvrossum If you can supply me with (error) logs I can check why the integration is not working for you. Otherwise you can uninstall it via HACS and the original integration will show up again.

corneyl, 2025-05-11:
Same issue here, I'd like to use this integration next to the 'official' one for controlling the power production. What's needed to change the domain? Is it only changing the `DOMAIN`  in `const.py`?

vincentwolsink, 2025-05-11:
@corneyl 

- `DOMAIN` in `const.py`
- `domain` in `manifest.json`
- Directory name

corneyl, 2025-05-12:
@vincentwolsink That works well indeed!

BassplayerPatrick, 2026-04-07:
@vincentwolsink if these are the things to change to have it run alongside the official integration:

- DOMAIN in const.py
- domain in manifest.json
- Directory name

at what point do I have to make these changes? I really don't want to break my current configuration.