vincentwolsink/home_assistant_enphase_envoy_installer #208: [BUG?] Entities update interval is 1 hour

LionKingNL, 2025-03-18:
**Describe the bug**
I'm not sure if it is a bug or a feature...
All entities update roughly once every hour. For the majority of the entities it is fine.
But as I'm using the "Total power production" entity to calculate my 'true' power usage (actual power usage = solar production - power export to grid), I would like to have update interval every minute (or 20 sec, that is the interval of my power meter).
(If I remember correctly, the official integration has a much shorter update interval)

**Account type**
- [ ] Installer
- [x] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [ ] Metered with CTs installed
- [ ] Metered without CTs
- [x] Standard

- FW version: D8.2.4382
- Amount of micro inverters connected: 10 x IQ8+

**Home Assistant**
- Version: 2025.3.3

**Additional context**
```
Relevant snippet of Home Assistant error log. 
```

vincentwolsink, 2025-03-18:
I would not set the minimum time between updates lower than 60s. The Envoy does not update the data that fast and this could also lead to requests stacking up and overloading the Envoy (especially with a lot of inverters connected).

But to give a educated answer on the issue I would need the debug logs.

catsmanac, 2025-03-18:
[Here's a long thread](https://github.com/briancmpbll/home_assistant_custom_envoy/issues/201) on intervals dropping to an hour, It's not HA integration related as the participants in the thread use another custom integration and the core integration.

LionKingNL, 2025-03-19:
@catsmanac, thanks for the link.
I'll have to be patient and wait for the issue to be fixed (by others).

closed 2025-03-19 by LionKingNL

vincentwolsink, 2025-03-21:
Confirmed in the debug log that fetching the data in the integration is working like it should. So it seems somehow the Envoy does not update somewhere internally.

LionKingNL, 2025-03-26:
After consulting the Enphase support (support_nl@enphaseenergy.com), it turned out that the polling-configuration (on Enphase's side) was changed to 1 hour. The support employee changed this to 5 minutes and it is working now.
I get entity updates every 5 minutes.