vincentwolsink/home_assistant_enphase_envoy_installer #209: [BUG] Actualisation Rythm changed for Inverters

apct69, 2025-03-20:
Hi, i wanted to implement new code today and i noticed that the actualisation rate changed for some of my micros. (pretty sure it was OK before since i had some curve drawed.)

On enlighten both type report every 15min, but not on HA: m215 15min, IQ7 1h:

[image]

**Account type**
- [ ] Installer

**Envoy**
- [ ] Metered with CTs installed

- Amount of micro inverters connected: 8

Thanks a lot :) 
(reloaded the integration already)

vincentwolsink, 2025-03-21:
See also #208 btw.

apct69, 2025-03-23:
Hi, indeed it also impacts other metrics like temperature or current.. I will take a look at #208 
Thanks

vincentwolsink, 2025-03-31:
If you contact Enphase support they should be able to fix it for you by fixing the update interval.

closed 2025-03-31 by vincentwolsink