vincentwolsink/home_assistant_enphase_envoy_installer #210: [BUG] Unable to initialise after Envoy upgrade to 8.3.5027

Kudos50, 2025-03-30:
**Describe the bug**
One of my Envoys has recently been updated to the latest firmware supporting a mix of IQ7 and IQ8. This Envoy has such a mix! After the upgrade I now get: Failed setup, will retry: Expecting value: line 1 column 1 (char 0)

**Account type**
Installer

**Envoy**
Standard

FW version: D8.3.5027
- IQ7: 8
- IQ8: 1

**Home Assistant**
- Version: Core 2025.3.4, Supervisor 2025.03.3, Operating System 15.0, Frontend 20250306.0
- Integration version: 0.6.7
 

**Relevant snippet of Home Assistant error log**
2025-03-30 12:22:50.283 INFO (MainThread) [custom_components.enphase_envoy.envoy_reader] FETCHING ENDPOINT endpoint_info TOOK 0.2533 seconds
2025-03-30 12:22:50.284 ERROR (MainThread) [custom_components.enphase_envoy] Unexpected error fetching envoy Envoy 122226007183 data
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 380, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 281, in _async_update_data
    return await self.update_method()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/__init__.py", line 107, in async_update_data
    await envoy_reader.get_data()
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 1251, in get_data
    await self.update_endpoints()  # fetch all remaining endpoints
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 1235, in update_endpoints
    self.data.set_endpoint_data(endpoint, getattr(self, endpoint))
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 349, in set_endpoint_data
    self.data[endpoint] = response.json()
                          ~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/httpx/_models.py", line 832, in json
    return jsonlib.loads(self.content, **kwargs)
           ~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
           ~~~~~~~~~~~~~~~~~~~~~~~^^^
  File "/usr/local/lib/python3.13/json/decoder.py", line 345, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
               ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/json/decoder.py", line 363, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
2025-03-30 12:22:50.294 DEBUG (MainThread) [custom_components.enphase_envoy] Finished fetching envoy Envoy 122226007183 data in 3.768 seconds (success: False)

vincentwolsink, 2025-03-31:
Hi! That seems like the Envoy might be returning invalid json. Could you send me the full error log so I can find which endpoint it fails on?

Kudos50, 2025-03-31:
[home-assistant_enphase_envoy_2025-03-30T10-22-51.305Z.log](https://github.com/user-attachments/files/19536020/home-assistant_enphase_envoy_2025-03-30T10-22-51.305Z.log)

Kudos50, 2025-03-31:
Request URL: https://192.168.1.123/info.xml
Request Method: GET
Status Code: 200 OK
Remote Address: 192.168.1.123:443
Referrer Policy: strict-origin-when-cross-origin

HTTP/1.1 200 OK
Server: openresty/1.17.8.1
Date: Mon, 31 Mar 2025 14:42:56 GMT
Content-Length: 2275
Connection: keep-alive
Strict-Transport-Security: max-age=63072000; includeSubdomains
X-Frame-Options: DENY
X-Content-Type-Options: nosniff

vincentwolsink, 2025-03-31:
Should be fixed in [0.6.8](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/releases/tag/0.6.8)

PR #212 (2025-03-31): Handle missing content-type in response headers
Fixes #210
closed 2025-03-31 by vincentwolsink