vincentwolsink/home_assistant_enphase_envoy_installer #211: [BUG] Today's Production and Consumption reset at 2am after daylight saving

twoenter, 2025-03-31:
**Describe the bug**
This morning I saw the Today's meters were not reset as I expect. 

The meters reset at 2am this morning instad of 00:00. 

The production meter also did not reset to 0, but appr. 10kWh 

**Account type**
- [ ] Installer
- [x] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [x] Metered with CTs installed
- [ ] Metered without CTs
- [ ] Standard

- FW version: D8.2.4264
- Amount of micro inverters connected: 8

**Home Assistant**
- Version: 2025.2.4

**Additional context**
```
Relevant snippet of Home Assistant error log. 

![Image](https://github.com/user-attachments/assets/16c670fb-75f1-4b22-9482-c4f6c95e6c05)

mnederlof, 2025-03-31:
I have the same, and fixed this by creating a utility meter helper entity for this. Not ideal but it is what it is. It just resets at 00:00 UTC which is common in programming.

I do not think it is wise for this integration to forcibly set the values to 0 between 00:00 and 04:00 for example (as the issue lies with enphase, not with this integration, and this integration just returns the value listed by the envoy)

So if you really want this fixed, i'd suggest reaching out to enphase support. I would expect they have some sort of fix, otherwise i'd think it would result weird resets in other parts of the world (for example Australia or US).

catsmanac, 2025-03-31:
Seeing the same on my non-metered Envoy. Wonder if this is Daylight Saving Time change artefact. Last year on April1, with older firmware, I had a 1 hour shift only on that day. In Mar 2023 also a 1 hour timeshift.

twoenter, 2025-03-31:
> I have the same, and fixed this by creating a utility meter helper entity for this. Not ideal but it is what it is. It just resets at 00:00 UTC which is common in programming.
> 

Did you fix it with a utiliy meter from the lifetime production sensor?

closed 2025-04-10 by vincentwolsink