vincentwolsink/home_assistant_enphase_envoy_installer #213: [FEATURE] Send production to mqtt on envoy standard

broekieman, 2025-04-06:
**Describe the feature**
A clear and concise description of what you would like.

I have an envoy standard and a Victron home battery. 
Normally you need the metered version and CT to push production to the Victron by modbus/mqtt. 

I saw a writing from enphase to make this work also on the standard. 

Can you make work? 

**Account type**
- [ x] Installer
- [ ] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [ ] Metered with CTs installed
- [ ] Metered without CTs
- [ x] Standard

- FW version: D7.xxx
- Amount of micro inverters connected: x

**Home Assistant**
- Version: 2023.x

vincentwolsink, 2025-04-06:
Hi @broekieman, can you point me to this writing?

broekieman, 2025-04-06:
[teb-00199-en-emea-ac-coupling-with-victron-battery-inverters-using-modbus-tcp-ip_4 (1).pdf](https://github.com/user-attachments/files/19622702/teb-00199-en-emea-ac-coupling-with-victron-battery-inverters-using-modbus-tcp-ip_4.1.pdf)

riemers, 2025-04-06:
i just bumped into this issue, i read the pdf. It talks about metered only, not standard though. In theory your right it could also work on standard to just show the PV production. But your installer needs to install the correct firmware and then you can just scan on your victron the PV enphase device. I don't believe you need anything from here. I know for metered setups that https://github.com/mr-manuel/venus-os_dbus-enphase-envoy exists. But i like this new idea better.

vincentwolsink, 2025-04-07:
@broekieman That document only covers the Envoy Metered. But actually this modbus (not mqtt) connection is something between the Envoy and Victron directly (as @riemers already stated).

Home Assistant also has modbus integrations, maybe you could craft something yourself. But this is out of the scope of this integration.

closed 2025-04-10 by vincentwolsink