vincentwolsink/home_assistant_enphase_envoy_installer #217: [BUG] Envoy production switch not working

Pakast, 2025-05-01:
**Describe the bug**
When toggling the switch.envoy_xxx_production switch, it is not working. When toggling in the Enphase Installer Toolkit, the value in Home Assistant does not update.

**Account type**
- [x] Installer
- [ ] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [x] Metered with CTs installed
- [ ] Metered without CTs
- [ ] Standard

- FW version: D8.2.4382
- Amount of micro inverters connected: 30

**Home Assistant**
- Version: 2025.4.3

**Additional context**
Log:

This error originated from a custom integration.

Logger: custom_components.enphase_envoy
Source: helpers/debounce.py:137
integration: Enphase Envoy (Installer) (documentation, issues)
First occurred: 1:06:23 PM (1 occurrences)
Last logged: 1:06:23 PM

Unexpected exception from <bound method DataUpdateCoordinator.async_refresh of <homeassistant.helpers.update_coordinator.DataUpdateCoordinator object at 0x7f09c82d08a0>>
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/debounce.py", line 137, in _handle_timer_finish
    await task
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 356, in async_refresh
    await self._async_refresh(log_failures=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 479, in _async_refresh
    self.async_update_listeners()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 178, in async_update_listeners
    update_callback()
    ~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 559, in _handle_coordinator_update
    self.async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1023, in async_write_ha_state
    self._async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1148, in _async_write_ha_state
    self.__async_calculate_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1089, in __async_calculate_state
    if extra_state_attributes := self.extra_state_attributes:
                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/sensor.py", line 474, in extra_state_attributes
    int(value), tz=datetime.timezone.utc
    ~~~^^^^^^^
TypeError: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'

vincentwolsink, 2025-05-01:
Can you please supply the diagnostics file for the integration?

Pakast, 2025-05-01:
[config_entry-enphase_envoy-01JT58M7D08YFG2DZQ605CTACA.json](https://github.com/user-attachments/files/19999903/config_entry-enphase_envoy-01JT58M7D08YFG2DZQ605CTACA.json)

vincentwolsink, 2025-05-03:
Do you have inverters replaced at some point? Or inverters that are currently not functional?

Pakast, 2025-05-04:
I have registered 7 inverters that I did not yet install.

vincentwolsink, 2025-05-04:
You can try if [0.6.9-beta1](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/releases/tag/0.6.9-beta1) fixes it.

Pakast, 2025-05-05:
Yes, it is working in the 0.6.9-beta1.

closed 2025-05-11 by vincentwolsink