vincentwolsink/home_assistant_enphase_envoy_installer #223: [FEATURE] Add maxReportWatts

hapklaar, 2025-05-22:
**Describe the feature**
When checking the api on https://ipaddress/api/v1/production/inverters I see a max value for each panel is recorded. Could this also be added as an entity in HA?

```
  {
    "serialNumber": "482320000000",
    "lastReportDate": 1747915033,
    "devType": 1,
    "lastReportWatts": 374,
    "maxReportWatts": 374
  },
```

**Account type**
- [x] Installer
- [ ] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [ ] Metered with CTs installed
- [ ] Metered without CTs
- [x] Standard

- FW version: D8.3.5050
- Amount of micro inverters connected: 6

**Home Assistant**
- Version: 2025.5.2

vincentwolsink, 2025-05-22:
I can add it. I am not sure about the time window though. Some documentation states it is the maximum wattage the inverter delivered in its lifetime, but for me it shows a value lower than yesterday's high. So it seems to be per day.

vincentwolsink, 2025-05-23:
Available in 0.7.0

closed 2025-05-23 by vincentwolsink