vincentwolsink/home_assistant_enphase_envoy_installer #228: Unable to find the switch to power off the system

RemonWillems, 2025-07-01:
Hi,
Regarding the documentation, you should be able to shutdown the solar system but i can't find the switch. How can i enable the switch?

Thank you in advance!

Best regards,

Remon

rogervdh, 2025-07-05:
Have look at the Envoy device. You should see the production switch there (see below). The entity name of the switch is switch.production.

[image]

RemonWillems, 2025-07-05:
Hi,
You should expect that..:

[image]

vincentwolsink, 2025-07-05:
@RemonWillems Are you sure you have installer rights to your Enphase system? If you don’t have the rights the button does not show up.