vincentwolsink/home_assistant_enphase_envoy_installer #232: Today’s Energy Consumption - too high value

miketenwolde, 2025-07-18:
Today’s Energy Consumption sensor is showing a too high value.

Enphase app is showing correct value

Today’s Energy Production is showing correct value

[image]

vincentwolsink, 2025-07-18:
Hi @miketenwolde. The values are coming from the Envoy directly. The integration does not do any manipulation.

Unfortunately the daily counters from the Envoy are a bit buggy. The app uses the cloud platform that has its own method of keeping statistics.

You could wait and see if the value is correct tomorrow, otherwise you could try to restart the Envoy.

miketenwolde, 2025-07-18:
Thanks I will wait and otherwise reboot