vincentwolsink/home_assistant_enphase_envoy_installer #235: Envoy xxxxxxxxx Production/ firmware Version / grid profile became unavailable everyday at 23:01

GentleJan, 2025-07-25:
Every day at 23u01 I get an Logbook entry that de Envoy xxxxx (name of my envoy) Production / Firmware Version and GridProfile sensors became unavailable. I dont know is it is my envoy/IQ gateway self of the plugin that is misreading something. Maybe it is a default night reset of the envoy/gateway? But it is not nice, because minute later it will "turn on" and that will trigger automations at my site because of the "change" of the switch.
I am not doing anything with the device around that time!

Latest envoy firmware (D8.3.5167)
hardware 800-00553-r02 (non-metered envoy)

vincentwolsink, 2025-07-25:
Hi @GentleJan, this is known behavior from the Envoy.

Easiest fix for you would be in your automations to ignore the status change of the switch if the previous state was unknown or unavailable.

GentleJan, 2025-07-25:
Hi @vincentwolsink ,

Thanks for you reaction.
I was just monitoring it for a week now with an automation because i let Frank Energie "control" my PV/Enphase. First i was thinking that they were sending a cloud API call that was disturbing the Enphase. But find out that is wasn't them. So i thought it was the plug -in or envoy itself. 
But you "say" it is the envoy itself that makes it unavailable...  So i will make the workaround in my automation

vincentwolsink, 2025-07-25:
Connections to the Envoy API timeout during that time. Seems like it is doing a planned restart (since it is the same time each day).

Other Enphase integrations suffer from it too. Not much I can do 🙁